exsilium/pxbee-trigger #4: Trouble connecting to SmartThings

stevenlovegrove, 2018-02-28:
I'm trying to reproduce your results with my SmartThings hub but I'm completely new to both smartthings handlers and zigbee. I followed your instructions to set

#define ZCL_MANUFACTURER    "Leviton"
#define ZCL_MODEL           "ZSS-10"

in custom.h in order to try to avoid having to write a custom handler. I have built and uploaded the new firmware and I am greeted with the devices interactive prompt if I open a serial terminal to it. Is there anything else I need to do to add the device from within my smartthings app? Do I need to initiate pairing or something?

Can you recommend any useful way of debugging zigbee / smartthings?

Thanks again for your help!

exsilium, 2018-02-28:
Hi @stevenlovegrove !

1. SmartThings expects certain XBee settings to have been set:

https://github.com/exsilium/pxbee-trigger/blob/2573c08c45ed26d318066fa75e1fb5bb1953326e/include/custom.h#L9

 you can either do it via XCTU (after putting the radio into Bypass mode) or, as you have connected to the module via serial, you can use the `2` key for *Init additional radio settings* to do that for you. Makes sense to power-cycle the module after the settings have been set.

2. From the SmartThings phone app, press the ´+`  to add new *things*, the ZSS-10 should be found.

stevenlovegrove, 2018-02-28:
Hmm, I'd tried using the 'Init additional radio' previously but that didn't seem to help.

To set the XBee PAN, I assume the easiest way is through XCTU? I tried to use `./fwup -d /dev/... -B` to get into bypass mode, but the tool hangs after writing 'Entering bypass handling!'. If I leave it open then XCTU can't open the port, and if I ctrl-c fwup then XCTU asks me to reset the device, and then when I do it times out.

Also, this is probably a silly question, but how would I discover the PAN of existing devices on my network? Is the PAN likely to be the same as the 'Network Device Id' listed on api.smartthings.com for one of my zigbee devices?

Thanks again for your help on this, I really appreciate it!

exsilium, 2018-02-28:
Ctrl-c from fwup, but when adding to XCTU, add it as a standard non-programmable module. Don't reset the module as it will exit from the bypass. See if that helps.

Previously the PAN ID was available when logging in via https://graph.api.smartthings.com/
Currently I see only 16-bit PAN ID presented there and a channel... I wonder why they changed this.. 🤕

exsilium, 2018-02-28:
Actually the EUI listed, **is** the PAN ID via the https://graph.api.smartthings.com/

exsilium, 2018-03-03:
I just checked the XBee radio settings, and the PanID does not match the EUI listed nor the `${hub.zigbeeId}`. So I currently can't tell from where it would be easy to read the PAN ID that SmartThings creates... :(

exsilium, 2020-07-17:
This is an old and stale issue. I've since migrated the modules from SmartThings Hub v2 to Hub v3 - the only connectivity isssue I had is that the module must be in close proximity to the coordinator. Initial pairing and discovery does not work via routers, even in the node join time is infinite (0xFF). After successful pairing with SmartThings, further links are established within the mesh with other routers. Closing.

closed 2020-07-17 by exsilium